esp8266/esp8266-wiki #26: make[1]: xtensa-lx106-elf-gcc: Command not found

SpaceTeddy, 2015-02-12:
followed your wiki instruction almost 1:1.
the only change is that I used the 0.9.5 SDK

compiling the IoT Demo failed always with the above error. how can I fix it.
I use Ubuntu 14.04.1 in Fusion VM.
thx

khalilst, 2016-06-08:
You need to clone the git repository **esp-open-sdk**.

  git clone --recursive https://github.com/pfalcon/esp-open-sdk
  make STANDALONE=y |& tee make0.log
  export PATH="$HOME/esp-open-sdk/xtensa-lx106-elf/bin/:$PATH"
  alias xgcc="xtensa-lx106-elf-gcc"

See more detailed information at below link.
http://www.esp8266.com/wiki/doku.php?id=toolchain

mano1979, 2017-04-04:
in my case, i cannot compile the sdk either.

markh2016, 2018-06-06:
There are three videos showing you how to set this up I made yesterday owing to problems with git hub repositories   ,,,  bad files  that are downloaded  and causing this not to compile 

These are on my Facebook home page 

Please see the following and there after you shouldn't, touch wood have any further problems , fingers crossed  

Hope this helps as  I also had extreme problems getting this to even compile as of yesterday I  also spent some three hours and more finding the answers although there are are other with similar bad experiences with this how ever I managed to solve this 

You might like to view this yourselves 

Here  is the address for you  ( if you follow this I think  you will have at  least some success ) 

https://www.facebook.com/mark.harrington.142892/videos/2091296477783030/?t=0
https://www.facebook.com/mark.harrington.142892/videos/2091321701113841/?t=28
https://www.facebook.com/mark.harrington.142892/videos/2091378541108157/?t=0

Azjhong, 2020-04-09:
I encountered the same problem, and I solved it as follows.
First, you can refer to the solution provided by [espressif](https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started/windows-setup.html).
I was caused by the lack of msys32 installation. After the installation was successful, the above problems were successfully solved.