esp8266/esp8266-wiki #31: Wiki memory map additions

linagee, 2015-04-07:
Not sure how to contribute to the wiki, so I'm opening an issue.
On the Wiki memory map: https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map

0x60009A00 is a "wifi memory interface".

Proven by: libphy.a disassembled (IDA).
Found 0x60009A00 as a static string in use by these symbol objects:
(there may be more 0x6000 0000 addresses for wifi.)

set_rx_gain_testchip_50
do_noisefloor_lsleep_v50
do_noisefloor
start_dig_rx
stop_dig_rx
ram_get_noisefloor
ram_set_noisefloor
ram_start_noisefloor
sdt_on_noise_start
chip_v6_set_chan_rx_cmp
phy_dig_spur_prot
ant_switch_init

(I very much knew what they did. Still learning about this chip. I hope to extract RSSI from regular packets, not just when doing an AP scan.)

jcmvbkbc, 2015-04-07:
Please be more specific.
> 0x60000200: chip_v6_rxmax_ext_dig

the function uses offset 0x390 effectively accessing 0x60000590

linagee, 2015-04-07:
I was trying to make it concise. I think at this point I will need to start putting it into a spreadsheet so the data is not all over. ;)

jcmvbkbc, 2015-04-07:
You can open an issue for each region you'd like to add. You can start a thread at esp8266.com or anywhere you see fit, that might be easier and it may have wider audience.