esp8266/esp8266-wiki #32: Can't contribute like Wikipedia

nerdralph, 2015-04-10:
Is there an easy way to contribute to the wiki?  I don't see an option to edit like there is on most other wikis.

camilstaps, 2015-04-12:
See https://help.github.com/articles/about-github-wikis/. Wikis basically work as any other content on GitHub.

But yes, this is not Wikipedia, so it works a bit differently. (It's better.)

nerdralph, 2015-04-12:
That doesn't explain how to contribute to a wiki that isn't for one of your
own repositories.  Your loss; I'll just put the info on my blog.  Not
better than Wikipedia when people don't contribute...

jcmvbkbc, 2015-04-12:
The only way I know of is asking https://github.com/themindfactory to give you editor access. Collaborators can't do that.

egbit, 2015-04-12:
There's also a new wiki now for ESP8266, based on DokuWiki:
http://www.esp8266.com/wiki/doku.php

pfalcon, 2015-04-23:
> Your loss; I'll just put the info on my blog.  Not better than Wikipedia when people don't contribute...

I fully agree. I've kept my additions in a local clone of this wiki for couple of months, now I just made a github fork and pushed it there: https://github.com/pfalcon/esp8266-wiki/wiki (Don't forget to use "Clone this wiki locally" link, as github wiki are better used as git clones of course).

Wiki and and project should be run by people who really contribute to it, like @jcmvbkbc , @tommie, @zarya. Not the likes of @themindfactory whose the only contribution is claiming that *they* are the community.

> There's also a new wiki now for ESP8266, based on DokuWiki: http://www.esp8266.com/wiki/doku.php

This is hilarious. There's no need for other wiki run by god knows whom god knows where. Here on github there's a good wiki, which everyone can fork and continue if self-proclaimed community masters disappear (or even before that). http://www.mt6260.com/ also says there's wiki at http://www.mt6260.com/wiki , but just the same there's no MT6260 wiki there, because it's all at http://www.kosagi.com/w/index.php?title=Fernvale_Main_Page , where people who really do stuff add content.

pfalcon, 2015-06-13:
@jcmvbkbc : See http://richard.burtons.org/2015/06/12/esp8266-cache_read_enable/ for references to few more SPI ROM mapping registers. You probably could filter them thru your knowledge and add them.