esp8266/esp8266-wiki #33: Toolchain is broken

steowens, 2015-04-16:
None of the examples build.  Make returns the following errors:
```
make[1]: xt-xcc: Command not found
make[1]: *** [.output/eagle/debug/obj/at_baseCmd.o] Error 127
make[1]: Leaving directory `/opt/Espressif/ESP8266_SDK/at_v0.20_on_SDKv0.9.3/user'
make: *** [.subdirs] Error 2
```

bneumann, 2017-01-06:
I am just fiddeling around with an ESP8266 and have the same problems. Seems that the repo is abandoned?

nzhome, 2017-01-07:
abandoned is not a word associated with github,  just fork it, and continue the work on it :) there are 239 other forks, see if they have made any fixes, otherwise hopefully someone like yourself can pick it up where others have left off.