esp8266/esp8266-wiki #38: Please enable editing of wiki pages

valentt, 2015-06-08:
Hi, would like to contribute by editing wiki pages but I don't see a way how to do it now.

I have forged wiki repository but there are no WIKI documentation pages there to edit. Can you please clarify how wiki documentation pages can be edited. 

Thanks.

camilstaps, 2015-06-08:
Duplicate of #32.

kentaylor, 2016-05-17:
It looks like this needs to be enabled as documented at https://help.github.com/articles/changing-access-permissions-for-wikis/ which would be useful as https://github.com/esp8266/esp8266-wiki/wiki/Hardware_ESP8266-Versions#esp8266 says "If you have any information describing the differences, please add it here!" but it is not possible.