esp8266/esp8266-wiki #41: osapi.h: error: implicit declaration of function

tiestvangool, 2015-11-15:
Hi experts,

When running the script as outlined in the esp8266-wiki/wiki/Building section, I get the following error: "implicit declaration of function" in osapi.h.
I receive this error on all clean Lubuntu VMs I've created.
Has anyone seen this before and/or guide me into a direction to get this resolved?

Thanks,
-Tiest

codetiestvangool@localhost:~$ ./blinky.sh
Cloning into 'source-code-examples'...
remote: Counting objects: 107, done.
remote: Total 107 (delta 0), reused 0 (delta 0), pack-reused 107
Receiving objects: 100% (107/107), 20.63 KiB | 0 bytes/s, done.
Resolving deltas: 100% (28/28), done.
Checking connectivity... done.
CC user/user_main.c
In file included from user/user_main.c:2:0:
user/user_main.c: In function 'user_procTask':
/opt/Espressif/ESP8266_SDK/include/osapi.h:12:21: error: implicit declaration of function 'ets_delay_us' [-Werror=implicit-function-declaration]
#define os_delay_us ets_delay_us
^
user/user_main.c:34:5: note: in expansion of macro 'os_delay_us'
os_delay_us(10);
^
user/user_main.c: In function 'user_init':
/opt/Espressif/ESP8266_SDK/include/osapi.h:31:25: error: implicit declaration of function 'ets_timer_disarm' [-Werror=implicit-function-declaration]
#define os_timer_disarm ets_timer_disarm
^
user/user_main.c:51:5: note: in expansion of macro 'os_timer_disarm'
os_timer_disarm(&some_timer);
^
/opt/Espressif/ESP8266_SDK/include/osapi.h:32:24: error: implicit declaration of function 'ets_timer_setfn' [-Werror=implicit-function-declaration]
#define os_timer_setfn ets_timer_setfn
^
user/user_main.c:54:5: note: in expansion of macro 'os_timer_setfn'
os_timer_setfn(&some_timer, (os_timer_func_t )some_timerfunc, NULL);
^
/opt/Espressif/ESP8266_SDK/include/osapi.h:30:31: error: implicit declaration of function 'ets_timer_arm_new' [-Werror=implicit-function-declaration]
#define os_timer_arm(a, b, c) ets_timer_arm_new(a, b, c, 1)
^
user/user_main.c:60:5: note: in expansion of macro 'os_timer_arm'
os_timer_arm(&some_timer, 1000, 1);
^
user/user_main.c:63:5: error: implicit declaration of function 'system_os_task' [-Werror=implicit-function-declaration]
system_os_task(user_procTask, user_procTaskPrio,user_procTaskQueue, user_pr
^
cc1: all warnings being treated as errors
make: ** [build/user/user_main.o] Error 1
[/code]

by2coffee, 2015-11-16:
This is because Espressif "forgot" to provide the declaration of many SDK functions.

Can you try this:
1. Create a folder called `include` in `source-code-examples/blinky`
2. Download [espmissingincludes.h](https://github.com/Spritetm/libesphttpd/blob/master/include/espmissingincludes.h) into  `source-code-examples/blinky/include`
3. Add `#include espmissingincludes.h` before the line `#include "osapi.h"` in the file `source-code-examples/blinky/user/user_main.c`
3. Try to build again

tiestvangool, 2015-11-16:
by2coffee,

Thanks for taking a stab at this! I followed your instructions and the declaration issues seem to be gone  but a got a bunch of new ones in return :-)
Any thoughts on the error messages below?

CC user/user_main.c
In file included from user/user_main.c:2:0:
include/espmissingincludes.h:4:1: error: expected identifier or '(' before '<' token
 <!DOCTYPE html>
 ^
In file included from user/user_main.c:2:0:
include/espmissingincludes.h:7:19: error: character constant too long for its type [-Werror]
     <meta charset='utf-8'>
                   ^
include/espmissingincludes.h:13:5: error: stray '\302' in program
     <title>libesphttpd/espmissingincludes.h at master · Spritetm/libesphttpd 
     ^
include/espmissingincludes.h:13:5: error: stray '\267' in program
include/espmissingincludes.h:13:5: error: stray '\302' in program
include/espmissingincludes.h:13:5: error: stray '\267' in program
In file included from user/user_main.c:2:0:
include/espmissingincludes.h:100:32: error: missing terminating ' character [-Werror]
       <form accept-charset="UTF-8" action="/Spr
                                ^
include/espmissingincludes.h:100:7: error: missing terminating ' character
       <form accept-charset="UTF-8" action="/Spr
       ^
In file included from user/user_main.c:2:0:
include/espmissingincludes.h:290:28: error: character constant too long for its type [-Werror]
   <form accept-charset="UTF-8" action="/users/s
                            ^
In file included from user/user_main.c:2:0:
include/espmissingincludes.h:290:752: error: data definition has no type or storage class [-Werror]
 UTF-8" action="/users/set_protocol?protocol_selector=subversion&amp;protocol_ty
                                                                     ^
include/espmissingincludes.h:290:752: error: type defaults to 'int' in declaration of 'protocol_type' [-Werror=implicit-int]
include/espmissingincludes.h:290:766: error: 'clone' undeclared here (not in a function)
 "/users/set_protocol?protocol_selector=subversion&amp;protocol_type=clone" clas
                                                                     ^
include/espmissingincludes.h:290:771: error: expected ',' or ';' before string constant
 rs/set_protocol?protocol_selector=subversion&amp;protocol_type=clone" class="in
                                                                     ^
include/espmissingincludes.h:290:834: error: invalid suffix "bf546db17ea3e18aadc64af2bac413ffc1a098" on integer constant
 clone" class="inline-form js-clone-selector-form " data-form-nonce="27bf546db17
                                                                     ^
include/espmissingincludes.h:290:937: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 1a098" data-remote="true" method="post"><div style="margin:0;padding:0;display:

by2coffee, 2015-11-19:
Sorry, my link was pointing at the github page for that file, not the header file itself. Please try this [espmissingincludes.h](https://raw.githubusercontent.com/Spritetm/libesphttpd/master/include/espmissingincludes.h)

tiestvangool, 2015-11-26:
Installed lower version of the software package.

closed 2015-11-26 by tiestvangool

QinYUN575, 2018-05-10:
You're attempting to compile C++ with a plain C compiler. Try g++ instead of gcc.