esp8266/esp8266-wiki #42: fatal error: ets_sys.h: No such file or directory  #include "ets_sys.h"

tiestvangool, 2015-12-13:
Experts,

As I run the blinky script I get the fatal error: ets_sys.h: No such file or directory
 #include "ets_sys.h".

I don't know how to resolve the and the internet is not very helpful. Anyone who has seen this or knows how to interpret this issue?

I run a clean install of Lubuntu and/or GalliumOS

Much appreciated
-Tiest

by2coffee, 2015-12-14:
Could you please confirm if you have updated `SDK_BASE` in the Makefile, to point to directory where the SDK is present?

joppo, 2016-05-16:
Same here, the SDK_BASE points to where it's supposed to (at least as far as I know).
[screenshot](https://onedrive.live.com/redir?resid=D8AF97C913A4809!120159&authkey=!ALSyg845vcmDAu8&v=3&ithint=photo%2cPNG)

GanesanGuru, 2018-03-16:
Hi fellas,

Can anyone explain how to overcome this " fatal error: ets_sys.h: No such file or directory" error??
My $PATH is /opt/xtensa-lx106-elf/bin:/home/esp8266/Desktop/ESP_OPEN_SDK/esp-open-sdk/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games