esp8266/esp8266-wiki #57: make fails on x64 debian8

Karmek, 2016-11-07:
Hi There,

I've pulled a copy of toolchain on my debian:
```
uname -a
Linux vm-debian 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt25-2+deb8u2 (2016-06-25) x86_64 GNU/Linux;
```
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.6 (jessie)
Release:	8.6
Codename:	jessie
```
when i try to run makem i get the following error:
```
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: Entering directory '/home/dennis/Projekte/ESP8266 ESP-01/github_SDK_repo/esp-open-sdk/crosstool-NG'
../Makefile:155: warning: overriding recipe for target '/home/dennis/Projekte/ESP8266'
../Makefile:126: warning: ignoring old recipe for target '/home/dennis/Projekte/ESP8266'
../Makefile:162: warning: overriding recipe for target '/home/dennis/Projekte/ESP8266'
../Makefile:155: warning: ignoring old recipe for target '/home/dennis/Projekte/ESP8266'
../Makefile:181: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
../Makefile:176: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
configure: WARNING: you should use --build, --host, --target
configure: WARNING: invalid host type: ESP-01/github_SDK_repo/esp-open-sdk/crosstool-NG
checking build system type... Invalid configuration `ESP-01/github_SDK_repo/esp-open-sdk/crosstool-NG': machine `ESP-01/github_SDK_repo/esp-open-sdk/crosstool' not recognized
configure: error: /bin/bash scripts/config.sub ESP-01/github_SDK_repo/esp-open-sdk/crosstool-NG failed
../Makefile:143: recipe for target '_ct-ng' failed
make[1]: *** [_ct-ng] Error 1
make[1]: Leaving directory '/home/dennis/Projekte/ESP8266 ESP-01/github_SDK_repo/esp-open-sdk/crosstool-NG'
Makefile:140: recipe for target 'crosstool-NG/ct-ng' failed
make: *** [crosstool-NG/ct-ng] Error 2
```
Any Idea what might be going wrong?

Regards,

Karmek

Karmek, 2016-11-07:
Nevermind this one. I was careless enough to leave a space in one of the folder names. The scripts did not like it much ;-) However now i get some permission problems:

sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
/bin/bash: .config: Permission Denied
../Makefile:130: recipe for target '_toolchain' failed

Maplicant, 2017-05-18:
Getting the same permission problem as you, how did you resolve it?