esp8266/esp8266-wiki #66: Connecting ESP8266 to FRITZ!Box 7490

tobau, 2017-09-13:
Hello,

i got a strange Problem:
I've written a C-Code which connects to an AP and serves TCP at some ports. It works totally fine in my Network (Hitron Hub). A few days ago, i've moved to my parents place to install an ESP to do some lightning control. 
I've changed the SSID and the WPA2 Key in my Code. I've Copy and Pasted both, so its impossible that there is a typo...

How Ever: When i'm powering up the ESP, it finds the network, and tries to connect to it, but it doesn't work. The Log-File of the FRITZ!Box 7490 says, that there was the wrong key. 

Here is the Debug-Out from the ESP:

>scandone
no FRITZ!Box - Intern found, reconnect after 1s
reconnect
scandone
no FRITZ!Box - Intern found, reconnect after 1s
reconnect
scandone
no FRITZ!Box - Intern found, reconnect after 1s
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt
state: 5 -> 2 (2a0)
rm 0
reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt

and this repeats the whole time.

Is it a problem with TKIP/AES/CCMP? Which encryption does the ESP support? 
Is there any way to get a succesfull connection?

Thank you very much!
Greetings

evaluationevaluation, 2018-01-21:
Hi,

had a similar issue.
In my case it was the "connection mode" of the ESP.
        wifi_station_set_config(&stationConf);
        wifi_set_phy_mode(PHY_MODE_11B);
        wifi_station_set_auto_connect(1);
        wifi_station_connect();
The second line seem to force ESP to use 802.11B for connection with the access point and this mode has been deactivated in my WiFi environment. Setting the access point to accept 802.11b/g/n solved the issue for me.
Don't know if this second line is really needed or if it would be better simply to remove this line. Perhaps ESP then connects with the best mode available?!

Greetings
 Eva

tobau, 2018-01-22:
Hello Eva,

 

thanks for your Reply!
Currently i'm working on my Diploma Thesis and haven't enough time to try it out. But i will do so! :) 

Thank you,

greetings

 

Tony

 

Gesendet: Sonntag, 21. Januar 2018 um 12:41 Uhr